robberwick/tauradigm-teensy #45: readsensors update timing tied to logging packet send timing

markmellors, 2020-10-30:
autonomous functions may need sensor updates more frequently, which could cause sensor read duplication

robberwick, 2020-10-30:
[this comment](https://github.com/robberwick/tauradigm-teensy/issues/52#issuecomment-719663717) is pertinent here I think:

> I think we should do a test to find the maximum update rate of the IMU and the ESC. Technically the standard update rate for the ESCs is 50Hz, so no need to read encoders more frequently than that for PID purposes. Current 100Hz (10ms logging loop) is fine, though we may get more speed precision if we read them more slowly.
Odometry could benefit from being updated as fast as we can read the IMU. Which could be once per loop, though again, I think 50-100Hz is probably fine.
I guess none of the above changes whether it's once per loop or not. Depends if we go for a standardised loop time or put everything on its own timer

robberwick, 2020-10-30:
So what do we want to do here then? 50Hz updates for Encoders, IMU and TOFs? Or should we go for different update schedules for each?

robberwick, 2020-10-31:
thinking about it, it costs us nothing to have separately defined intervals for each of those things, so I'll go ahead and define a 50Hz interval for all 3, and we can always tweak them individually, or even consolidate them into one call later if we decide that's the way to go.

markmellors, 2020-10-31:
I think that's a good idea. I think 50hz for IMU & encoders makes sense. We're currently doing nothing with the ToF, and when we do, it'll be for localisation that will probably run at ~10Hz, so I'd start with that. 10Hz logging too, I've not needed the higher rate for anything yet.

commit 68838219 (2020-10-30): Update sensors every loop

Fixes #45
commit d8d33ea0 (2020-10-31): Update sensors every loop

Fixes #45
PR #60 (2020-10-31): define seperate timers for updates and logging

Define seperate timers for all the various updates and
logging operations

Fixes #45
closed 2020-10-31 by markmellors