robberwick/tauradigm-teensy #52: Move encoder updating logic to HAL

robberwick, 2020-10-30:
the encoder updating logic is currently in the super loop, and we should probably move it into the HAL. perhaps we should consider the updating of them once per loop as part of #45 and #46

markmellors, 2020-10-30:
I think we should do a test to find the maximum update rate of the IMU and the ESC. Technically the standard update rate for the ESCs is 50Hz, so no need to read encoders more frequently than that for PID purposes. Current 100Hz (10ms logging loop) is fine, though we may get more speed precision if we read them more slowly. 
Odometry could benefit from being updated as fast as we can read the IMU. Which could be once per loop, though again, I think 50-100Hz is probably fine.
I guess none of the above changes whether it's once per loop or not. Depends if we go for a standardised loop time or put everything on its own timer

robberwick, 2020-10-30:
yeah, that's true. I'll stick to just moving the logic for instantiating and updating the encodor readings to the HAL for this PR, as that's orthogonal to those points, so no sense getting it mixed up with questions of frequency or timing.

commit 302d6fb1 (2020-10-30): Move encoder updating logic to HAL
Fixes #52
PR #59 (2020-10-31): Move encoder updating logic to HAL
Fixes #52
closed 2020-10-31 by markmellors