robberwick/tauradigm-teensy #63: battery low voltage behaviour kicks in too early and too aggressively

markmellors, 2020-10-31:
I think the low battery thereshold kicks in too aggressively. It often triggers during normal driving with an averagely charged battery. I don't know if we need to just lower the threshold or do some smoothing to filter out motor peak current blips/dips. The current warning level is fine as a warning (as it kicks in at about ~25% remaining capacity, 11.1V) but its annoying it also kills motor power when clearely there's enough capacity to keep driving

markmellors, 2020-10-31:
measured battery voltage drops about 0.2V when motors are stalled/wheelspinning

robberwick, 2020-10-31:
I had a TODO to debounce the battery voltage readings anyhow, because we're currently doing it every loop. perhaps one approach would be to check that battery voltage in a periodic fashion on the scale of seconds rather than milliseconds, and even then only act if the last `n` readings are below the minimum value.